simondlevy/BreezySLAM #1: Raspberry Pi ARM V6

robvoi, 2015-02-04:
Hi,

do you think there is a chance thy python SLAM runs on a raspberry pi?

Robert

gunthercox, 2015-02-04:
I am also interested in this. I would be willing to test this but I'm not sure what hardware is required, aside from the Raspberry Pi.

@robvoi Any suggestions?

simondlevy, 2015-02-04:
Pi is too slow for real-time SLAM: http://home.wlu.edu/~levys/scanrates.png
I recommend spending another $30 and to get an ODROID U-3.

Of course, you'll also need a Lidar unit.  I used the Hokuyo URG-04LX ($1150), but you could also buy a Neato XV-21 for $320, or just its RPLidar unit for $400.  Another BreezySLAM user reports good results with the RPLidar.

robvoi, 2015-02-05:
Thanks for the feedback. I have a RP-Lidar. So this point is set.

Two more questions:

1. May it be possible to run the python SLAM with a lower update rate (like 1Hz) on a Pi. Just as test.
2. What do you think about the Raspberry Pi 2 to run SLAM?

simondlevy, 2015-02-05:
1. No, you'll likely get aliasing, unless the vehicle is moving very slowly.
2. The Pi2 has a quad-core ARM-7 processor, so you should get significant speedup, but I haven't tested it.