simondlevy/BreezySLAM #3: Matlab Breezyslam

ALHarake, 2016-05-27:
Hello,
when running the matlab logdemo.m file I receive this error (MATLAB has encountered an internal problem and needs to close)

Could you please point out what might be the problem ?

it is very weird because this is the first time I get such an error!

Prati11, 2017-03-21:
Hi,
when running this logdemo('exp2', 1) I received an error "The specified superclass 'WheeledRobot' contains a parse error, cannot be found on MATLAB's search path, or is shadowed
by another file with the same name."

Please help

muratsalar, 2018-04-03:
I have the same error can you help?

dbyrash, 2019-07-05:
>> logdemo('exp2', 1)
Error: File: logdemo.m Line: 42 Column: 1
Function definitions are not permitted in this context. 

Please help me with this. This error is occurring in Matlab. 

Also, 
>> logdemo('exp2', 1)
Undefined variable "WheeledRobot" or class "WheeledRobot.MinesRover".
Error in logdemo (line 66)
    robot = WheeledRobot.MinesRover();

Do I need to add something to the code?