simondlevy/BreezySLAM #6: hokuyo_connect:_connect:_get_sensor_info: response does not match:

joelmaranhao, 2016-06-30:
# Environment

- Using LiDAR URG-04LX (Hokuyo Automatic Co.,Ltd.)
- Python 2.7.6

```
~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty
```

```
$ uname -a
Linux mycomputer 4.2.0-27-generic #32~14.04.1-Ubuntu SMP Fri Jan 22 15:32:26 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

# Error

This error isn't systematic.
If I try enough times it will eventually work (?)

```
~$ ./BreezySLAM/examples/urgslam.py
breezylidar.URG04LX:hokuyo_connect:_connect:_get_sensor_info: response does not match:
breezylidar.URG04LX:hokuyo_connect:_connect:_get_sensor_info_and_params: unable to get status!
breezylidar.URG04LX:hokuyo_connect:_connect: could not get status data from sensor
Traceback (most recent call last):
  File "./BreezySLAM/examples/urgslam.py", line 36, in <module>
    lidar = Lidar(LIDAR_DEVICE)
TypeError: Wrong/insufficient arguments passed to URG04LX.__init__() 

```

joelmaranhao, 2016-07-06:
May be we don't have the same firmware, could you check?
```
~/BreezyLidar/examples$ python urgtest.py 
```

```
===============================================================
Model:    URG-04LX(Hokuyo Automatic Co.,Ltd.)
Firmware: 3.3.01(23/Jun./2008)
Serial #: H1002589
Protocol: SCIP 2.0
Vendor:   Hokuyo Automatic Co.,Ltd.
===============================================================
```

simondlevy, 2016-07-06:
Yes, you're right, Joel.  You are running firmware 3.3.01, whereas mine is
3.4.03:

levy@kern:~/Desktop/BreezyLidar/examples$ ./urgtest.py

===============================================================
Model:    URG-04LX-UG01(Simple-URG)(Hokuyo Automatic Co.,Ltd.)
Firmware: 3.4.03(17/Dec./2012)
Serial #: H1320770
Protocol: SCIP 2.0
Vendor:   Hokuyo Automatic Co.,Ltd.
===============================================================
Iteration:   1: Got 682 data points
Iteration:   2: Got 682 data points
Iteration:   3: Got 682 data points
Iteration:   4: Got 682 data points
Iteration:   5: Got 682 data points
Iteration:   6: Got 682 data points
Iteration:   7: Got 682 data points
Iteration:   8: Got 682 data points
Iteration:   9: Got 682 data points
Iteration:  10: Got 682 data points
10 scans in 0.906507 seconds = 11.031354 scans/sec