simondlevy/BreezySLAM #8: Can I run BreezySLAM from pcap Lidar data.

nickponline, 2016-07-26:
I have a .pcap file of LIDAR data captured from VLP-16. Can I run BreezySLAM on it?

simondlevy, 2016-07-26:
Not as its stands.  But if you're interested, I can try to write some Python code that'll parse the pcap and run it through BreezySLAM.

nickponline, 2016-07-26:
That would be really useful. I can provide you will a pcap file if that helps ..

simondlevy, 2016-07-26:
BreezySLAM only requires a list of millimeter scan distances.  It knows how to interpret these because you first instantiate a class representing the parameters of your Lidar (scans per second, readings per scan, scan angle, etc.)  If you want, you can also pass odometry values (forward and rotational speed), but it's best to avoid those at first.  This example shows how it works with a Hokuyo URG04LX:

  https://github.com/simondlevy/BreezySLAM/blob/master/examples/urgslam.py

The Velodyne VLP-16 may be providing more data per scan that BreezySLAM knows how to handle.  I'll fool around a little with the PCAP file you posted, to see whether this is the case.

simondlevy, 2016-07-26:
So this PCAP stuff turns out to be nontrivial.  This guy's post looks promising, but I was unable to follow it through to an actual usable Python code sample: http://binarymillennium.blogspot.com/2008/08/more-python-pcap-with-pcapy.html

There's also an ROS driver that will read PCAP files containing Velodyne output: http://wiki.ros.org/velodyne_driver

If you can get the PCAP data into a format where there is a sequence of distance values for each successive scan, we can try and write a new BreezySLAM class for the VLP-16.  Otherwise I suggest you look into ROS.

ALHarake, 2016-07-27:
Hello nickponline, I would like to know what results you get using a velodyne lidar, if you please. I have been using low cost lidars (rplidar and another home-made-one), and it seems that when the number of points per scan increases, it becomes harder for the algorithm to converge. 
For example how many points you have per scan, and if it is possible to send an image of the generated map it would be great.

nickponline, 2016-07-27:
@ALHarake going to work on this over the next few days and then can respond with some data.

ALHarake, 2016-07-27:
@nickponline great, I'm eagerly waiting the results as buying a velodyne lidar isn't possible.

Sanatramesh, 2018-03-08:
@simondlevy I am also working with VLP-16 LiDAR. Could you let me know how to write a Laser class for VLP-16, i.e, how do I specify the no of beams and their verticle angle?

@nickponline were you able to get breezySLAM working with VLP-16?