simondlevy/BreezySLAM #17: Using a previously generated map for localization (Python)

JackToaster, 2017-11-07:
I would like to use BreezySLAM for localization (no mapping) in a known map.
In python, the the algorithm classes have the `getmap(mapbytes)` method which returns a bytearray with the map, but it seems there is no way to do the opposite (set the map to a bytearray).
Once the map is set, I would also like to be able to localize in the map without modifying the map, but I can't find a way to do this either.

simondlevy, 2017-11-07:
Thanks for the suggestions, Jack.  I should be able to get to it this weekend.

simondlevy, 2017-11-12:
I've added a a **setmap()**  method, and a **should_update_map** flag (default=True) to the update() method. I don't have data that would test these modifications, so let me know if you run into difficulty.

closed 2017-11-12 by simondlevy

JackToaster, 2017-11-16:
I just did a few simple tests, and it looks like it works. Thanks!