simondlevy/BreezySLAM #19: Odometry doesn't improve SLAM.

epinal, 2018-01-05:
Hi Simon,
I really need your help to understand why the odometry data is not improving at all the SLAM.

I've been using breezySLAM for over a year and I've always had the issue that when the robot is moving between long hallways the SLAM gets messed up, the robot position doesn't change but the map moves instead. I suppose that since the laser scans (the slam image) are almost the same the algorithm doesn't know that the robot is moving. So I thought that this will improve using the odometry data.
Based on your WheeledRobot and the C++ source code I implemented this class which I already tested and works like a charm, meaning that if I move forward 10 feet I get dxy_millimeters = 3040 and if I rotate to the right 90 degrees I get dtheta_degrees = 90 to 94 so I know I'm getting the right dxy_millimeters, dtheta_degrees, dt_seconds values.

    `class Robot:

    def __init__(self, wheel_radio: float, half_axle_length: float, ticks_per_turn: int,
                 communicator: ArduinoCommunicator):
        """
        Class to compute odometry in a wheeled robot with encoders
        :param wheel_radio: wheel radius in mm
        :param half_axle_length: half distance between wheels axle in mm
        :param ticks_per_turn: encoders ticks per turn
        :param communicator: Arduino communicator to get encoders count
        """
        self.wheel_radio = wheel_radio
        self.half_axle_length = half_axle_length
        self.ticks_per_turn = ticks_per_turn
        self._communicator = communicator

        self._odo1_previous_count = None
        self._odo2_previous_count = None
        self._timestamp_seconds_prev = None

        self._request = OdometryRequest().to_json()

    def extract_odometry(self):
        """
        Function to get odometry data from the Arduino and convert it to degrees
        :return: tuple (timestamp, odo1_degrees, odo2_degrees)
        """
        timestamp = None
        odo1_degrees = None
        odo2_degrees = None

        response = self._communicator.send_and_receive(self._request)
        if response:
            try:
                odo_resp = OdometryResponse(response)
                odo1_degrees = self.ticks_to_degrees(odo_resp.odometry1)
                odo2_degrees = self.ticks_to_degrees(odo_resp.odometry2)
                timestamp = odo_resp.timestamp / 1000  # data comes in milliseconds

            except Exception as e:
                LOGGER.exception(e)

        return timestamp, odo1_degrees, odo2_degrees

    def compute_velocities(self):
        """
        Computes forward and angular velocities based on odometry.

        Returns a tuple (dxy_millimeters, dtheta_degrees, dt_seconds)

          dxy_millimeters forward distance traveled, in millimeters
          dtheta_degrees change in angular position, in degrees
          dt_seconds     elapsed time since previous odometry, in seconds
        :return:
        """
        dxy_millimeters = 0
        dtheta_degrees = 0
        dt_seconds = 0

        timestamp_seconds_curr, current_odo1_degrees, current_odo2_degrees = self.extract_odometry()

        if self._timestamp_seconds_prev and timestamp_seconds_curr:
            delta_odo1 = current_odo1_degrees - self._odo1_previous_count
            delta_odo2 = current_odo2_degrees - self._odo2_previous_count

            dxy_millimeters = self.wheel_radio * (math.radians(delta_odo1) + math.radians(delta_odo2))
            dtheta_degrees = (self.wheel_radio / self.half_axle_length) * (delta_odo2 - delta_odo1)
            dt_seconds = timestamp_seconds_curr - self._timestamp_seconds_prev

        # Store current odometry for next time
        self._timestamp_seconds_prev = timestamp_seconds_curr
        self._odo1_previous_count = current_odo1_degrees
        self._odo2_previous_count = current_odo2_degrees

        return dxy_millimeters, dtheta_degrees, dt_seconds

    def ticks_to_degrees(self, ticks: float):
        """
        Convert encoder ticks to degrees
        :param ticks: float encoder ticks
        :return: float degrees
        """
        return ticks * (180. / self.ticks_per_turn)`

Im calling RMHC_SLAM.update(scan_mm, velocities) and Im getting the same messed up SLAM like if it wasn't taking into account that the robot is moving as shown in the velocity.
I can't upload a video where you see the map moving and the robot position stationary.
Let me know what can I do.

simondlevy, 2018-01-06:
I can't follow what you're doing here.  You don't say anything about the kind of vehicle you're using, what Lidar you're using, etc.  Also,  why did you look at the C++ code, instead of just sub-classing the existing Python [WheeledVehicle](https://github.com/simondlevy/BreezySLAM/blob/master/python/breezyslam/vehicles.py#L24-L33) class?

epinal, 2018-01-06:
First of all closing the ticket isn’t helpful at all, you didn’t give the chance to answer your questions. So thanks for being so polite and helpful. I’m going to answer them anyway:- You don’t need what kind of vehicle I’m using, business copyright , I’m telling you that the odometry and the convertion to velocities is fine but the slam doesn’t improve. - You don’t need to know what Lidar I’m using, business copyright, I’m telling you that the Lidar works perfect and the slam works pretty well until the robot get into places where the Lidar scans are almost the same. - I looked into the C++ code which is 100 times better than your python code because I needed to understand how it works and transform encoders ticks to radians and i found it in C, C++ and even Matlab but not in Python. Now you have your questions answered and either way doesn’t help at all. Don’t worry I’ll figure it out and maybe I’ll fork your repo rewrite the code following Python Coding style and start helping people so they don’t have to deal with you being rude. Regards,Ernesto PinalSoftware ArchitectAnimusoft Corporation - power the extraordinary2196 NW 82nd AVE Doral, FL 33122 | Office : 786-224-4599 ext 707 |www.animusoft.com | www.alive.ai ---- On Fri, 05 Jan 2018 19:45:05 -0500  Simon D. Levy<notifications@github.com> wrote ----I can't follow what you're doing here.  You don't say anything about the kind of vehicle you're using, what Lidar you're using, etc.  Also,  why did you look at the C++ code, instead of just sub-classing the existing Python WheeledVehicle class?

epinal, 2018-01-06:
Thanks for closing the issue without giving me the chance to answer your questions. So helpful...
I'll figure it out.

simondlevy, 2018-01-06:
Sorry, Ernesto -- I didn't mean to close the issue; I must've accidentally
hit the wrong button!

epinal, 2018-01-06:
LOL, Ok no problem.
Im using a SweepLidar http://scanse.io/ on a SuperDroidLT2 https://www.superdroidrobots.com/shop/item.aspx/lt2-tracked-atr-robot-platform/1513/
The lidar and the encoders are working as expected. I've been able to navigate buildings using the slam map and position.
I debugged the code and slam.update(scan, velocities) is being called with the right data and Im getting the map but without any improvement.
What do you think that I can check to make sure the velocities are being used by the SLAM algorithm?

simondlevy, 2018-01-06:
Your best bet is probably to look at the C code internals, specifically, the functions [scan_update](https://github.com/simondlevy/BreezySLAM/blob/master/c/coreslam.c#L431-L458) and  [scan_update_xy](https://github.com/simondlevy/BreezySLAM/blob/master/c/coreslam.c#L232-L255).  If you can put some debugging printf() statements in there, you may be able to see what's going on.

epinal, 2018-01-06:
Ok. I’ll take a look on Monday and I’ll let you know my findings. Have a good weekend. 
Thanks.

epinal, 2018-01-08:
Hi Simon,
I took a look into the C code and it seems to be fine.
I got a question here [CoreSlam update](https://github.com/simondlevy/BreezySLAM/blob/master/python/breezyslam/algorithms.py#L105-L116)
- What is the purpose of `#Update velocities`?
- Why update velocities if  `self.velocities` is used in `self._scan_update` which was already called before and `self._updateMapAndPointcloud(velocities, should_update_map)` uses the original `velocities`.

simondlevy, 2018-01-08:
I wrote this code in a few years ago, so I can't be very confident in answering  :^(  But I believe that the #Update velocities section is how new odometry evidence gets introduced into the algorithm, AFTER the scan has been computed based on the existing evidence.  

Specifically, the dt seconds in velocities[2] (time elapsed since previous reading) is turned into a frequency (velocity_factor), which then converts the linear (xy) and angular (theta) differences into rates (mm/sec, degrees/sec)  Then the new velocity evidence is available to the point-cloud computation. Something like:

1. Update scan based on current odometry.
2. Update odometry.
3. Update point-cloud based on scan from (1) and updated odometry from (2)
4. Goto 1.

I'm pretty sure I based this on the approach in the original [article](https://pdfs.semanticscholar.org/1c7e/7af4388b17b137badc6ec19e9724e6bd91e4.pdf), but again, it's been a long time!

epinal, 2018-01-08:
Ok, So following your logic:
- 1.Update scan based on current odometry.
- 2.Update odometry.
- 3.Update point-cloud based on scan from (1) and updated odometry from (2)
- 4.Goto 1

We should change [this](https://github.com/simondlevy/BreezySLAM/blob/master/python/breezyslam/algorithms.py#L115-L116) to:
 `self._updateMapAndPointcloud(self.velocities, should_update_map)`
Am I right?

simondlevy, 2018-01-08:
Yeah, I see your point: what I've implemented in the code doesn't follow the logic I just laid out.     

Looking at the C++ code, I see that it's consistent with the Python code, but not with the logic.  So there must've been some reason why I wrote the code that way, but at this point I can't even recall.  I just spent some time looking at the core C code as well, and still can't make sense of why I used the two different velocity computations as I did.

Unfortunately our winter term has just started, and I'm buried in teaching and other projects.  Rather than confusing things further, I'll suggest you try making that change and seeing whether it improves things.  I wish I could be more helpful ....

epinal, 2018-01-08:
LOL, I agree, I checked the C and C++ code too and I couldn't figure out why you did it that way that's why I asked. I'll make the change and test again. I know you are busy so thanks for helping me.

epinal, 2018-01-16:
Hi Simon,
It did improve changing [this](https://github.com/simondlevy/BreezySLAM/blob/master/python/breezyslam/algorithms.py#L115-L116) to:
 `self._updateMapAndPointcloud(self.velocities, should_update_map)`
You can close the issue.
Thanks for your help.

simondlevy, 2018-01-16:
Cool, thanks, I've changed it as you suggest.

closed 2018-01-16 by simondlevy