simondlevy/BreezySLAM #23: About setmap() function error

lihp6207, 2018-04-17:
Hi, Simond :
I have the same issue with "https://github.com/simondlevy/BreezySLAM/issues/17".
But when I use the function setmap(). ex: slam.setmap(mapbytes)
The following error appeared:
------------------------------------------------------------------------------------------------------------
  File "/usr/local/lib/python2.7/dist-packages/breezyslam/algorithms.py", line 130, in setmap
    self.map.set(mapbytes)
AttributeError: 'pybreezyslam.Map' object has no attribute 'set'
------------------------------------------------------------------------------------------------------------
It seems cannot find pybreezyslam.Map.set().
Did I have a wrong usage?
Thank you!

andrewda, 2018-07-25:
Running into the same issue. Any updates?

simondlevy, 2018-07-25:
Make sure you've downloaded the latest version of BreezySLAM and are running Python 3.  I've stopped supporting Python 2.

andrewda, 2018-07-25:
Ah gotcha, that'd be the problem. I was using Python 2.

simondlevy, 2018-07-25:
There could still be a problem -- if so, I'll be happy to look into it.

andrewda, 2018-07-26:
Just got home and was able to test it with Python 3 -- seems the issue is still present, same error as before.

lihp6207, 2018-07-26:
Hi, simondlevy and andrewda!
I have solved the problem not long ago by using slam.map.__init__ function instead of slam.setmap.
EX:
    slam = breezyslam.algorithms.RMHC_SLAM(lidar, MAP_SIZE_PIXELS, MAP_SIZE_METERS)
    slam.map.__init__(MAP_SIZE_PIXELS, MAP_SIZE_METERS, mapbytes)

You can try it !
Thanks!

andrewda, 2018-07-26:
Thank you, @lihp6207, `slam.map.__init__()` worked! Definitely still a plus to get the `set()` method working properly, but thanks again for the fix in the meantime!

lihp6207, 2018-07-26:
It's my pleasure!

simondlevy, 2018-07-26:
Thanks for figuring this out, lihp6207!  I've fixed slam.setmap() and tested it, but I'd appreciate it if you guys would give it a try and let me know if it's still not working.

andrewda, 2018-07-26:
Works like a charm, appreciate it!

closed 2018-07-26 by simondlevy