simondlevy/BreezySLAM #31: use option -std=c99 or -std=gnu99 to compile your code

joelmaranhao, 2018-07-13:
Getting the follow issue while compiling

```
root@qb57:~/BreezySLAM/python# sudo python setup.py install;
running install
running build
running build_py
copying breezyslam/sensors.py -> build/lib.linux-x86_64-2.7/breezyslam
copying breezyslam/__init__.py -> build/lib.linux-x86_64-2.7/breezyslam
copying breezyslam/vehicles.py -> build/lib.linux-x86_64-2.7/breezyslam
copying breezyslam/algorithms.py -> build/lib.linux-x86_64-2.7/breezyslam
running build_ext
building 'pybreezyslam' extension
x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c pybreezyslam.c -o build/temp.linux-x86_64-2.7/pybreezyslam.o
pybreezyslam.c: In function 'Scan_update':
pybreezyslam.c:312:9: error: 'for' loop initial declarations are only allowed in C99 mode
         for (int k=0; k<PyList_Size(py_scan_angles_degrees); ++k)
         ^
pybreezyslam.c:312:9: note: use option -std=c99 or -std=gnu99 to compile your code
pybreezyslam.c:348:5: error: 'for' loop initial declarations are only allowed in C99 mode
     for (int k=0; k<PyList_Size(py_lidar); ++k)
     ^
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

Found this thread on [stackoverflow](https://stackoverflow.com/questions/24881/how-do-i-fix-for-loop-initial-declaration-used-outside-c99-mode-gcc-error)

may be that helps resolving.

joelmaranhao, 2018-07-13:
Resolved by adding suggested flag

```python
module = Extension('pybreezyslam',
    sources = SOURCES,
    extra_compile_args = ['-std=gnu99'] + SIMD_FLAGS + OPT_FLAGS
    )

```

simondlevy, 2018-07-13:
Thanks, Joël.  I have added this line to setup.py.

closed 2018-07-13 by simondlevy