simondlevy/BreezySLAM #33: number of samples with RP Lidar A2

kenkyusha, 2018-08-15:
I have been experimenting with RP Lidar A2 and with the default settings of 200 and 250 samples (minimum samples for updating the map) I do not get any map built. If I lower the rate to 50 or 100 I get something really fussy and not really good.
Is this problem with python interface or could it be hardware problem?
Currently the Lidar is connected over USB port.

andrewda, 2018-08-15:
Same here. The highest number of samples I can get reach before the map stops being built is 120, even on my Jetson TX1 which is surprising as I'd expect it to be able to handle stuff like this. What's more, the CPU never goes above ~50% while generating the map, so I'm not sure where the bottleneck could be. I can delve more into system information and setup when I get home in a few hours.

simondlevy, 2018-08-15:
Yeah, I and others have had a some difficulty using these RP Lidar units with BreezySLAM, having to tune that min-samples parameter.  I'm not sure whether the solution lies in re-writing the [acquisition](https://github.com/SkoltechRobotics/rplidar) library or the core C in BreezySLAM.  Given how [fast](https://raw.githubusercontent.com/simondlevy/BreezySLAM/master/breezyslam.png)  BreezySLAM already runs on even inexpensive single-board computers like ODROID using other lidar units (like the Hokuyo URGs), my approach would be to try and speed up the RP Lidar acquisiton code.

andrewda, 2018-08-15:
Alright, that makes sense. I'll look into making my own acquisition library in something faster than Python. Thanks!

kenkyusha, 2018-10-20:
Using the C++ is much better and you get much more points! Haven't tested the slam yet but I would guess it is good (looking at the offline data and map generation)