simondlevy/BreezySLAM #38: C++ Demo Examples not working

Meghadoot, 2019-01-02:
I have cd to BreezySLAM/cpp, and do 'sudo make install'. But when I then cd to BreezySLAM/examples and do 'make cpptest', (I have correctly set the LIBDIR paths in both cpp and examples folder's makefiles)
[image]
 
I got error as shown in snap. Can you give me some suggestion? Thanks in advance!

Meghadoot, 2019-01-02:
On the other hand, Pytest works without any issues(can be seen in following snap)

[image]

simondlevy, 2019-01-02:
From the instructions:

**You may also need to add the following line to your ~/.bashrc file:**

```export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib```

Meghadoot, 2019-01-02:
Yes I already have updated ~/.bashrc as per instructions.
still error persists ..

Meghadoot, 2019-01-02:
Thanks. Now resolved it.
Issue was with path variable.

closed 2019-01-02 by Meghadoot