simondlevy/BreezySLAM #72: mapbytes makes no sense although lidar scan is very reasonable

Avi-avidan, 2020-04-11:
hi, 
I am getting a nice scan from my Slamtec RPLIDAR A1 (see image)
[image]

LIDAR_DEVICE = '/dev/ttyUSB0'

dists[:10] = [-0.0, -0.0, -0.0, -640.6713808877225, -616.3971794359838, -605.3760240984459, -588.6830172771465, -573.8061655912974, -584.4876481055396, -581.9273308674532]

angles[:10] = [-0.0, -0.0, -0.0, -109.72776180811681, -97.11084997755711, -89.79905036668178, -79.29883460350986, -69.50168580266731, -64.17311907687618, -59.87972603044943]

I am initializing objects as follows -
LaserModel = RPLidar
slam = RMHC_SLAM(LaserModel(), MAP_SIZE_PIXELS, MAP_SIZE_METERS, max_search_iter=5)
viz = MapVisualizer(MAP_SIZE_PIXELS, MAP_SIZE_METERS, 'SLAM')
mapbytes = bytearray(MAP_SIZE_PIXELS * MAP_SIZE_PIXELS)

running this simple loop to update my map -
for i in range(5):
    time.sleep(0.1)
    scan_dict = simple_express_scan(ret_points=NUM_POINTS)
    dists ,angles = get_raw_data(scan_dict, ang_off=4)
    time.sleep(0.1)
    slam.update(dists, scan_angles_degrees=angles)
    time.sleep(0.1)
    x, y, theta = slam.getpos()
    print(x, y, theta)    
    slam.getmap(mapbytes)

print(x, y, theta) always outputs: 5000.0 5000.0 0.0 even when I move my robot around (with the lidar on top)

this is the image I get -
[image]

any advice? what am I doing wrong?

Avi-avidan, 2020-04-11:
this is the map I get after slam.update() with 10,000 points (200 at a time) -
[image]

any advice would be greatly appreciated. thanks in advance.

Avi-avidan, 2020-04-11:
this is the 10000 points when I show it with matplotlib - 
[image]

any advice will be appreciated. thanks.

simondlevy, 2020-04-15:
Have you tried the [sketch](https://github.com/simondlevy/BreezySLAM/blob/master/examples/rpslam.py) I already wrote for RPLidar A1?

Avi-avidan, 2020-04-16:
here are the prints after running python3 rpslam.py
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

(rpslam.py:20166): Gdk-CRITICAL **: 21:57:03.273: gdk_cursor_new_for_display: assertion 'GDK_IS_DISPLAY (display)' failed
Too many measurments in the input buffer: 711/500. Clearing buffer...
Too many measurments in the input buffer: 780/500. Clearing buffer...
Too many measurments in the input buffer: 778/500. Clearing buffer...
Too many measurments in the input buffer: 809/500. Clearing buffer...
Too many measurments in the input buffer: 784/500. Clearing buffer...
Too many measurments in the input buffer: 775/500. Clearing buffer...
Too many measurments in the input buffer: 771/500. Clearing buffer...
Too many measurments in the input buffer: 813/500. Clearing buffer...
Too many measurments in the input buffer: 781/500. Clearing buffer...
Too many measurments in the input buffer: 770/500. Clearing buffer...
Too many measurments in the input buffer: 784/500. Clearing buffer...
Too many measurments in the input buffer: 781/500. Clearing buffer...
Too many measurments in the input buffer: 770/500. Clearing buffer...
Too many measurments in the input buffer: 819/500. Clearing buffer...
Too many measurments in the input buffer: 784/500. Clearing buffer...
Too many measurments in the input buffer: 780/500. Clearing buffer...
Too many measurments in the input buffer: 782/500. Clearing buffer...
Too many measurments in the input buffer: 786/500. Clearing buffer...
Too many measurments in the input buffer: 783/500. Clearing buffer...
Too many measurments in the input buffer: 808/500. Clearing buffer...
Too many measurments in the input buffer: 784/500. Clearing buffer...

Avi-avidan, 2020-04-16:
I am ssh'ing from the terminal into the robot so display is unavailable.
any advice?

Avi-avidan, 2020-04-16:
I got a scan plotted running the [sketch](url) in jupyter notebook. after replacing viz.display with matplotlib.pyplot -

simondlevy, 2020-04-16:
Nice work, sorry for the trouble!

Avi-avidan, 2020-04-16:
no trouble. thanks again for this great repo and your timely support :)

Thuong-ironman, 2023-05-03:
Hi,
I am using rplidar a1 and have an issue: 
Too many bytes in the input buffer: 4010/3000. Cleaning buffer rplidar

Do you have any idea to solve this?

jackhou66, 2024-04-09:
> Hi, I am using rplidar a1 and have an issue: Too many bytes in the input buffer: 4010/3000. Cleaning buffer rplidar
> 
> Do you have any idea to solve this?

have you resolve this questions?